contiki-os/contiki #2676: Contiki modulation

soeun727, 2021-03-06:
hi i want to check cooja modulation.
My experiment environment is cc2420, then is that right that the cooja using O-QPSK modulation?

And also how can I get the graph Eb/No graph(the energy per bit to noise power spectral density ratio)? 
I want to check modulation and if it possible I want to change it.

Please give me answer. Thanks.

atiselsts, 2021-03-07:
Cooja does not simulate the physical layer.

The closest you can get to the physical layer is to use the `Msp802154BitErrorRadio` radio interface on Sky or Z1 nodes. It does simulate realistic bit error distribution in the received packets, assuming interference-free channel. But this is still part of the purely digital logic in the receiver, nothing from the analog domain is simulated in Cooja.

soeun727, 2021-03-07:
Thanks for giving comment. 
Could you told how can I change Msp802154BitErrorRadio radio interface to my simulation?

soeun727, 2021-03-07:
Thanks for giving comment.
Could you told me how I can change Msp802154BitErrorRadio radio interface to my simulation?
Do I have to change Makefile? 
I search about it, but there is not much information about it..

atiselsts, 2021-03-08:
If you open a Cooja simulation file (.csc) that is using msp430 motes, there will be one or more `<motetype>` definitions. By default this definition includes this line, defining the radio interface:

       <moteinterface>org.contikios.cooja.mspmote.interfaces.Msp802154Radio</moteinterface>

Change that to:

       <moteinterface>org.contikios.cooja.mspmote.interfaces.Msp802154BitErrorRadio</moteinterface>

You can also change the interfaces from the Cooja GUI.