contiki-os/contiki #2679: Unable to collect data from rpl-collect view

deepti948, 2021-04-12:
I am working on two way communication protocol, therefore to analyze my work I need data from collect view, so that I can compare my results and conclude on the same.
I was able to get results for in-built rpl collect program example but when I tried to execute the same for my own program I was unable to generate  results.
Can anyone please help me to know how to generate rpl-collect for my own program?

Libtissam, 2022-11-08:
Hi! I have the same issue, can you help me please?

hknaydin, 2023-02-25:
In this article, I will explain the collect-view example that I added today. A few days ago I had the opportunity to explain this in detail. I will only show the changes on makefile and udp-server, udp-client. The letters in red indicate the additions on the Makefile.

hello friends in this video I will show you the use of collect-view on cooja. First of all, let me state that, measurements are made using the samples available on the internet and the files udp-sender.c udp-sink.c.

https://yazilimdnyasi.wordpress.com/2019/10/19/collect-view-example-for-contiki-os/