contiki-os/contiki #2686: Incorrectly handling negotiated options of telnet servers

jerrytesting, 2021-08-08:
Hello,

In the implementation of telnet servers until version 3.0 and even the latest commit 32b5b17, telnet servers incorrectly handle negotiated options. 

According to the general constraints of [RFC854](https://datatracker.ietf.org/doc/html/rfc854), during negotiating some disabled command options or unnegotiated commands, telnet servers must give WILL/WONT or DO/DONT response for DO and WILL commands, respectively. 

However, telnet servers may not give any responses in this case. This bug appears as telnet servers put all responses in a fixed-length buffer in the implementation. Telnet servers only put messages into buffer but don't have a check whether successfully or not. Hence, when the buffer is full, it can lead to responses lost. This bug could lead to clients waiting forever and other effects.

Could you have a check? Thanks a lot.

OS-WS, 2021-09-06:
Hi, 
Is there any plan to fix this issue?

urlyy, 2024-08-20:
Excuse me, how can I reproduce this bug?