contiki-os/contiki #2696: Confusion regarding speed % of simulation?

umairjamali, 2021-10-29:
Hello everyone. This is my time with Cooja simulation. As I start my simulation (all nodes within each others' gray/green circles), the speed is around 10, 12% in first 9 seconds and suddenly we see speeds of more than 100% afterwards.

My question is that : what does that indicate really? How to interprete it? Thank you.

abhiverma866, 2022-05-12:
The speed of simulation completely depends on the speed of your processor and available RAM. We see time as per out clock but processor works in GHz (clock speed). That is why some times 10 min simulation may end within 1 min. Point to be noticed is that there is no effect on the results.

Laileet, 2023-04-04:
Any info how do i speed it up! or shall i keep it as it is?
Actually sometimes becomes 0 and whole experiment is not completed that's my problem.

Laileet, 2023-04-04:
> The speed of simulation completely depends on the speed of your processor and available RAM. We see time as per out clock but processor works in GHz (clock speed). That is why some times 10 min simulation may end within 1 min. Point to be noticed is that there is no effect on the results.

Any info how do i speed it up! or shall i keep it as it is?
Actually sometimes becomes 0 and whole experiment is not completed that's my problem.