contiki-os/contiki #2710: cc2538-cc2592-dk flashing issue

ceylanim, 2022-10-18:
Hi.

I bought a couple of cc2538-cc2592-dk for test.

[https://www.alibaba.com/product-detail/CC2538-CC2592-DK-Development-Board-for_1600456619006.html](https://www.alibaba.com/product-detail/CC2538-CC2592-DK-Development-Board-for_1600456619006.html)

I am not good at these boards. Before I bought, I have used cooja simulation software under contiki-ng.
There was cc2538 demo example on the boards. I didnt know how i can flash the board. I compiled the hello world example with ‘make TARGET=cc2538dk hello-world’ command and than I copied the hex file under build/c2538dk to windows. On windows side, I have used Smartrf flash programmer software. In the first test, I flashed that but disable bootloader was selected. I couldnt flash the same board again. It was failed.
After that, I have typed a code in hello-world.c . That was

`#define FLASH\_CCA\_BOOTLDR\_CFG\_ENABLE 0xF0FFFFFF
`

Compiled again, copied the hex file to windows side. And unselected the disable bootloader.

Before flashed I have unplugged the board and holding the select button, plugged again.

I have flashed the board. After that I was able to flash the board again. But that wasnt printing on serial. I tried to connect on putty, I couldnt see anything. I tried to flash the cc2538 demo, it was successfull, but there was nothing again. Even blinking…

My first question is about the first board. I couldnt flash it again. I think it is about bootloader. How can I flash it again? Maybe with jtag?

I have this jtag.

[image]

But I dont know how i do it. There is not a document for it.

Second question is why my methods are not working, why cant I see anything on board.

Is my method right?

What do you advice me?

mistergomes, 2022-10-18:
Dear Ceylanim,

Because Contiki supports the MCU Architecture doesn't mean that it also
supports the board configuration.
Supported original cc2538-dk boards may have the LEDs and other GPIO's
mapped on pins different from the dk you just bought.
Also your board has a power amplifier and to enable the PA (cc2592) you
also need to perform some configurations before you can use it right away.
Otherwise you won't see any packets being sent from the node.

If you were able to flash it with some tool/device, I believe that it was
correctly flashed, expect that you cant see anything because of the
possibilities mentioned above.
And by the way, that picture seems to me like a logic analyzer and not a
flashing tool.

Good luck with that!

Regards.

ceylanim, 2022-10-18:
Thank you for information. I think it will be difficult for me.

mistergomes, 2022-10-18:
I just remember that TI's have the same boards with the cc2538 + PA, and I
got them working a long time ago.
 If your boards are an exact "clone", it should work in the same way.

Check this thread:
https://github.com/contiki-os/contiki/issues/1023

You just need some modifications in the code to connect with the cc2592.